AlexmagToast/LinuxCNC_ArduinoConnector #3: mispelled INPUTS

snelweg, 2023-09-23:
Typo

/home/x/LinuxCNC_ArduinoConnector/LinuxCNC_ArduinoConnector.ino: In function 'void reconnect()':
/home/x/LinuxCNC_ArduinoConnector/LinuxCNC_ArduinoConnector.ino:625:25: error: 'Inputs' was not declared in this scope
     for (int x = 0; x < Inputs; x++){
                         ^~~~~~
/home/x/LinuxCNC_ArduinoConnector/LinuxCNC_ArduinoConnector.ino:625:25: note: suggested alternative: 'puts'
     for (int x = 0; x < Inputs; x++){
                         ^~~~~~
                         puts
/home/x/LinuxCNC_ArduinoConnector/LinuxCNC_ArduinoConnector.ino:626:7: error: 'InState' was not declared in this scope
       InState[x]= -1;
       ^~~~~~~
/home/x/LinuxCNC_ArduinoConnector/LinuxCNC_ArduinoConnector.ino:626:7: note: suggested alternative: 'state'
       InState[x]= -1;
       ^~~~~~~
       state

exit status 1

Compilation error: 'Inputs' was not declared in this scope

  #ifdef INPUT
    for (int x = 0; x < Inputs; x++){
      InState[x]= -1;
    }

AlexmagToast, 2023-09-23:
i think you should take a look at your code again. It seems like you have a synthax error in your config.

snelweg, 2023-09-23:
You misspelled #ifdef INPUTS
You wrote #ifdef INPUT

AlexmagToast, 2023-09-23:
oh you are right, sorry i didn't see it. 
Thank you!

closed 2023-09-23 by snelweg