AlexmagToast/LinuxCNC_ArduinoConnector #8: execv: No such file or directory

DeHandigeGasten, 2023-10-28:
Hi,
after install and trying to run "loadusr arduino-connector" in "halrun", I get the following error:
`halcmd: <stdin>:1: execv(arduino-connector): No such file or directory
`
The file is visible in /usr/bin, but somehow it doesn't seem to find the file.

`printncdhg@debian:/usr/bin$ ls -l arduino-connector 
-rwxr-xr-x 1 root root 18291 Oct 28 11:47 arduino-connector
`

linuxCNC version: 2.8.4
Running on Xfce 4.12

Any suggestions to what might be the problem? Thanks!

AlexmagToast, 2023-10-28:
i am not shure if i can help you with this problem. 
One idea that comes to mind is the following:
On which Python version is your system running? I've tested it once on LinuxCNC 2.8.4 but had to change the python version from #!/usr/bin/python3.9 to #!/usr/bin/python2.7.

Hope this helps. 
- Alex

DeHandigeGasten, 2023-10-28:
Thanks for answering so quickly! 
Sorry, forgot to mention, but I've changed the python line to 2.7 indeed.
The wierd thing is, I can run it manually with python without any problems, other files in the same directory are fine too, but somehow this file is not findable by `halrun`.

AlexmagToast, 2023-10-28:
the python LinuxCNC interface is based on this library: https://linuxcnc.org/docs/html/hal/halmodule.html
It states the python file should be installed in your $PATH. 
I doesn't make sense, but maybe it is something related to that?

Also have you tried turning it off and on again ? :D 

I am sorry but i honestly don't have any idea what seems to be the problem and how you could debug this.

DeHandigeGasten, 2023-10-28:
Hi, Thanks!
We have been trying some stuff too. 
I've no idea why this works but what we did:
We coppied the file of `hal_manualtoolchange`, emptied the file, coppied the contents of `arduino-connector` into the `hal_manualtoolchange` file and renamed the file to  `arduino-connector`. 
Somehow this works.

Still, no idea why, but it works. Maybe someone else might find this usefull or has a clue of why this works:)

Thanks!

AlexmagToast, 2023-10-28:
this is the weirdest fix i have ever seen, but glad it works now 👍

Tchefter, 2023-10-28:
(Hi Alex) Hi DeHandigeGasten,
I tried to reproduce the error, this occurs when the arduino-connector file is not present in /usr/bin/
Did follow the Installation Guide stepbystep?
I also run on LinuxCNC 2.8.4. (Debian10)

Regards Fritz

closed 2023-10-28 by AlexmagToast