AlexmagToast/LinuxCNC_ArduinoConnector #9: Incremental Naming of Inputs and Outputs

ModuloFS, 2023-10-28:
Hello Alex,

i am playing around with an STM32 and the STM32duino framework. So if i using for example the Pin PB7 as an output it is sometimes hard for me to get the resulting pin number and thus i have to search in the STM32duino framework for the resulting pin number to control this pin.

Because of this behavior, i would suggest to change this so that the input and outputs are named incremental by there creation. So for example in case of the Inputs it would be the index of the InPinmap[] array. 

If you like the idea i may provide a pull request to this topic.

AlexmagToast, 2023-10-28:
Hi, 
If i understand the problem correctly you have to research which "Dx" Pinnumber would represent PB7, right? 
What will you enter into the InPinmap[] array then? Do you still have to research the stm32duino pin for PB7 or can you enter PB7 directly?

ModuloFS, 2023-10-28:
I mean the output though the serial protocol. For example if i use PB7 as an Input i get something like I25:0 or I25:1 and for the second input PB8 for exapmle i get I15:0 or I15:1. For me it would be much less confusing if i get I0:0 or I0:1 for PB7 and I1:x for PB8 because i have defined them in this way.  So long story short, in the readInputs() function i would change sendData('I',InPinmap[i],InState[i]); to sendData('I',i,InState[i]);

AlexmagToast, 2023-10-28:
ah now i got it... thanks for the explanation.
I see why you would want it that way. For Arduino users however i believe it is easyer to just keep the pin number. 

I guess we could add a stm32 mode or something which would switch it for the inputs, outputs, PWM & Analog ins as you suggested. That way the "standard" remains but for usecases like yours its easy to switch. What do you think?

ModuloFS, 2023-10-28:
Yes i think this is a good idea because the encoder library has to be changed as well to work with the STM32.

AlexmagToast, 2023-10-28:
good to know

ModuloFS, 2023-10-28:
I think for the first shot i would name without mention the controller type because it is not especially stm32 related. Should i solve it with a #define statement? Do you have a good suggestion for the name? Maybe something like "IO_PROTOCOL_INCREMENTAL"?

AlexmagToast, 2023-10-28:
I will create a new branch for this until it works and we can merge it in main. 
yes, i think #define statements sounds perfekt. What do you think about #define INCREMENTAL_PIN_NAMES? 

I think we should create a new "advanced settings" sektion and put it there.

ModuloFS, 2023-10-30:
created pull request #13