AlexmagToast/LinuxCNC_ArduinoConnector #14: Using Encoder with STM32duino Framework

ModuloFS, 2023-10-30:
Hello,

if anybody wants to use the encoder functionality together with the STM32duino Framework you have to replace the Encoder library by this modified version:

https://github.com/ModuloFS/Encoder_STM32duino

You have to check by yourself if your used pins collide in the interrupt handler. For example in my case i am not able to use PA9, PA8 and PC9, PB8 at the same time as interrupt pins. So if you want to use two encoders You have to connect the first at PA9 and PA8 and the second for example at PB3 and PA10.

With the library mentioned above up to 6 encoders are possible.

AlexmagToast, 2023-10-30:
thanks, good to know.

EhsanRasouli1371, 2025-01-12:
If we want to use the blue pill board and stm32ino instead of Arduino boards, we will face problems in defining the names of the pins.
Can you change the program codes so that we can also use the blue pill?