AlexmagToast/LinuxCNC_ArduinoConnector #18: Add stepgens?

blazini36, 2023-12-29:
I replied about this in the LinuxCNC forum post. I have a use for some low speed remote stepgens. I have 3 small stepper motors at a remote location, there are actually several locations on this one machine where this is the case. It would not be practical to take stepgens off a Mesa card for this, they are at max 1khz and don't require a realtime response. They would just need to be accurate and have basic position feedback. Is it possible to add this to this project?

AlexmagToast, 2024-02-19:
Sorry for the late reply. Yes, we will eventually integrate it.
Can you help me by describing how this Project should integrate Stepgens to be useful for your Usecase? 
How would you ideally set it up in HAL?

blazini36, 2024-02-20:
Hi, thanks for responding.

I'm currently working on a control panel that will use alot of the features of this project. I haven't actually used this yet as I'm still working on the panel but I should be trying it soon.

I was requesting stepgens as a thought for another project I do that already runs on LinuxCNC. It uses remotely mounted cameras that move around. I'd been looking to incorporate motorized zoom lenses but controlling them is sort of an issue using a mfrs controller with it's own serial protocol and using 6+ stepgens off a Mesa card for tiny motors without much travel isnt really feasible. This project seems like its about halfway there as if I just stuck an Arduino compatible MCU near the lenses it wouldn't take much to make it work with this, if it had a stepgen function that is.

As for hal, it's tough to say. Ideally it would just be modeled after a cut down version of the software stepgen as most hardware projects use to use that as a template anyway.  Personally it would be fine with me if it only had a step+direction mode and took position and velocity commands with a position feedback. Obviously it would be limited in update rate on lower end MCUs but thats not really an issue with what I have in mind as long as it can run multiple stepgens.

AlexmagToast, 2024-02-20:
Wow what an amazing Idea! Thank you for your Input.
Not sure when it is working, but it will eventually be implemented! 
I'll leave a comment here and let you know, when the time has come.