AlexmagToast/LinuxCNC_ArduinoConnector #20: Ecoder, key matrix, and joystick issue

ShadeTechnik, 2024-02-28:
I'm having issues getting the encoders, and keymatrix working. The Joysticks I'm not too worried about because I'm not actually using them but I have tried to setup a config for a break out board I created which includes 2 joystick ports. It looks like there is some counter conflict between joysticks and encoders.

I created this break out board for a mega 2650:
[image]

I also created a 32 key matrix panel with LEDs. I had inquired if this schematic would work with ArduinoConnector in the post on the LinuxCNC forum:
[image]

This is how it connects to the Arduino
[image]

I'm not getting anything from the keymatrix. The pins show up in hal but nothing in the serial monitor.

I'm also not getting any response from any of my 4 encoders. The first encoder is the standard MPG wheel you can get anywhere, the other 3 are my own PCB for an EC11 but it's similar to any of the EC11 modules you can get off Amazon. Again, they show up in HAL but the counter never moves. The only thing I know works is digital inputs, haven't tested any outputs yet.

The Python file and Arduino firmware:
[LinuxCNC_ArduinoConnector.zip](https://github.com/AlexmagToast/LinuxCNC_ArduinoConnector/files/14434451/LinuxCNC_ArduinoConnector.zip)

AlexmagToast, 2024-02-28:
Yes, I implemented the Encoders and Joysticks to use the same code in some parts, because I never expected people to use both. I'll change that with the next update. 

The led Matrix setup looks good to me, make sure to uncomment //#define KEYPAD and //#define MULTIPLEXLEDS so the code is actually parsed while compiled. 

As for the Encoders: 
uncomment  this line in the .ino file and it should work: //#define QUADENC

blazini36, 2024-02-29:
Ahh thanks, not sure how I missed that. I also had the encoder array setup wrong in the ino. I'm obviously not one for code.

I got that corrected but the encoders and key matrix are not behaving properly. If I disable the LED mux, the matrix works a bit better, like half of the keys seem to register presses in hal but it's very flaky. Activating an LED seems to lock the whole thing up. The wiring between the Arduino BOB and the custom keypad is just a ribbon cable so I doubt there is a wiring problem.

I have 4 encoders set up, these are all AB type encoders. Ive checked them with a scope and they all put out reasonable quadrature. 3 of them use 2 interrupt pins, one uses 2 non-interrupt. 3 of them just constantly count for some unexplained reason but seem to reset for a moment when turned, then go right back to counting. the other one just moves a single digit.

The only atypical things I'm doing is a couple of the analog pins are used as digital. From what I understand Arduino analog pins are referenced as digital after the last digital. I'm using something like pins 56 and 59 which are actually A4 and A9 or something like that. My matrix and mux LED pins are not following a numbered order wired to the Arduino, this can be seen in the drawings. This is all done for shorter PCB routing, I wouldn't expect there to be an issue just figured I'd mention it in case of some bug. I'm going to try setting up basic firmwares to test these items individually outside of this project.

Another problem worth mentioning is I keep getting "pin not found" errors anytime I use the arduino pins in a hal file. These pins all show up in halrun. They also show up in halshow with LinuxCNC running, for whatever reason it seems like the pins are created too late to be caught in the initialization process. I think this is the first userspace component I've ever used but I don't think Ive seen this before.

ShadeTechnik, 2024-03-07:
Looks like I sometimes forget to switch accounts on when I comment on github......

The encoder issues appear to be mistakes I made in config, so disregard those.

No idea why the Python component generates the "pin not found" errors as I haven't seen that on other userspace components. These errors only appear when connections are made in hal files. I have found this can be worked around by adding [HAL] TWOPASS = ON to the ini file.

The LED and key matrix are fixed in a recent PR. This can be closed once that is accepted.

cofhal, 2025-10-02:
When I configure a key as in Pin with 0, they are effectively created as hal pins, but when checking if the keys related to those hal pins are pressed, they do not change state, which would indicate to me that they are not connected to those pins, hal. When testing from Arduino through the serial monitor, pressing any key shows that it is activated.