AlexmagToast/LinuxCNC_ArduinoConnector #23: Massive lag on Python side

blazini36, 2024-03-28:
After getting this going on a Teensy 4.1 I noticed a pretty large amount of lag with the hal pin value changes in LinuxCNC. It's very obvious when spinning an MPG wheel fairly fast, it takes the hal pin seconds to catch up. There's even an obvious lag just turning on an output from HAL.

When watching the serial output in a terminal there is 0 noticeable lag. Turning on an output is instant and I cannot outrun the counter update just by spinning an MPG by hand.

I noticed there is a V2 branch that looks like an extensive rewrite. Is this main branch going to be updated with fixes or is this going to be abandoned for V2?

AlexmagToast, 2024-03-28:
what does pretty large amount of lag mean in your case? 
Are you sure the lag happens in the arduino connector and not in Linuxcnc? 
If I select a small scale and a low axis speed, I can easily outrun the Axis movement with the MPG, since the maximum speed is limited. But in this case it is not the fault of the arduino connector. 

Turning on an Output from Hal takes "long", as the Python skript has to look at the pin and check it's state. You can adapt it by changing time.sleep(0.01) to something smaller. Deleting it will use a lot of CPU resources. 

In the V2 we try to target obvious downsides with the V1. Most config will happen in a single file and multiple Arduinos are supported simultaneously. 

If issues are found I do my best to fix it. Eventually V2 will surpass V1 and I guess at this Point People will not care about V1 anymore. Right now it seems V1 is running pretty stable,  but from time to time new bugs are found. 

How can I help you?

ShadeTechnik, 2024-03-29:
> what does pretty large amount of lag mean in your case? Are you sure the lag happens in the arduino connector and not in Linuxcnc? 

I can't make the distinction between Python and LinuxCNC since this is a userspace component that runs in LinuxCNC. I've never seen update lag on the LinuxCNC side. All I can say is working directly with the MCU from the serial terminal attached to the same port updates are instantaneous. Watching the HAL counter pin in LinuxCNC after spinning an MPG wheel  you can watch the counter continue to count for literally several seconds after you stop spinning it.

>If I select a small scale and a low axis speed, I can easily outrun the Axis movement with the MPG, since the maximum speed is limited. But in this case it is not the fault of the arduino connector.

Not sure I follow. I'm just watching the arduino connector hal pin directly, it's not connected to an axis so there's no scale. Not that it matters but the last test I did was on a Teensy 4.1 (@600mhz) with the USB serial set to 2Mbaud. There's no issue getting the serial data into the PC, there's some issue on the HAL-Python side

> How can I help you?

Not sure. I've been testing Arduino Connector obviously, but I can't actually put it to use, the lag is just too much on encoders and outputs. I haven't actually made a HAL config for it since I'll likely have to come up with a different solution. If it can't be fixed, it can't be fixed....I'm just posting the issue to make you aware.

Geekoid85, 2024-11-17:
Hello,
If someone is encountering the same issue, I found a way to fix it !
You have to lower this delay in the python script from 0.01 to something like 0.001 seconds.
[image]

Just to clarify, the issue was that the arduino.counter.0 hal pin was lagging behind when I spun the mpg handwheel too fast.
(Actually it was quite a lot, often more than a second as previously stated by @blazini36 )
Turns out the arduino and LinuxCnc weren't the problem, the python script just wasn't updating quickly enough for a 100ppr handwheel.

I am a git newbie, could you please implement this fix and close this issue so that other users don't encounter this issue ?

@AlexmagToast Many thanks for your work, it's truly a great addition to LinuxCnc !

AlexmagToast, 2024-11-17:
Thanks for the feedback. @Geekoid85 I will change it in the Projekt.

closed 2024-11-17 by AlexmagToast