AlexmagToast/LinuxCNC_ArduinoConnector #24: Option for an "Enable" button to enable/disable all controls on the Arduino side.

gdiciancia, 2024-06-23:
I wanted to propose an enhancement. The idea would be to setup a digital pin, that may or may not be exposed to LinuxCNC, and could be setup a Digital input, or Toggle input. 
The input would be used in the LOOP section, to ignore or run most the functions (except for outputs coming from LinuxCNC), depending on the status of pin.
The status of the input, could potentially influence the status LED (pin 13), with a different pattern (i.e. LED OFF).
Thank you for considering this enhancement.

KennethThompson, 2024-06-23:
What is happening where you want to reduce the number of features being
processed per loop? We have a 2.0 version still in the works and that may
already solve your issue perhaps.

gdiciancia, 2024-06-23:
I have a button that I want to use to disable the functions of the Controller/MPG (arduino side), to avoid accidental presses. The connection between Arduino and HAL in LinuxCNC is required anyway to avoid startup issues, but disabling the software functions on the Arduino side (without stopping the connection) should be possible.
Hope this clarifies, otherwise let me know.

KennethThompson, 2024-06-23:
> I have a button that I want to use to disable the functions of the Controller/MPG (arduino side), to avoid accidental presses. The connection between Arduino and HAL in LinuxCNC is required anyway [...]

Ah ok, I see.  That is an interesting edge case that I don't have a easy answer for.  Alex may have more insight into what version 1.0 could do.  But we do not have any current plans to support a feature like that in 2.0.  However, it could be a future item and is consistent with some ideas Alex and I were considering a while back- namely the ability to create rules/logic on the python side that can be used to create virtual pins. A virtual pin would probably handle your situation- but its a long ways off.

gdiciancia, 2024-06-23:
Ok, I understand. Thanks for the prompt feedback.

AlexmagToast, 2024-06-23:
You want to do this kind of logic in hal. In my Setup i wired my Enable input (arduino.din.30 in my case) to my signal mpg-enable and used this to enable jogging for the axis. (X&Z in my case). 

net mpg-enable axis.z.jog-enable axis.x.jog-enable <= arduino.din.30

gdiciancia, 2024-06-29:
> You want to do this kind of logic in hal. In my Setup i wired my Enable input (arduino.din.30) to my signal mpg-enable and used this to enable jogging for the axis. (X&Z in my case).
> 
> net mpg-enable axis.z.jog-enable axis.x.jog-enable <= arduino.din.30
> 
> LinuxCNC will ignore the jogging commands by the MPG as long as mpg-enable is not enabled.

I will try this. Thanks.

closed 2024-06-29 by gdiciancia