AlexmagToast/LinuxCNC_ArduinoConnector #25: will it run?

snelweg, 2024-07-20:
Have bought me a couple of these boards, KinCony KC868-A16:
[image]

There is documentation:

https://www.arduino.cc/reference/en/libraries/kc868-arduino-library/
https://www.kincony.com/forum/showthread.php?tid=1666

Question is will it work?

KennethThompson, 2024-07-20:
Dang, those are cool! Like Mesa esp32 boards! Alex can comment on
compatibility- but these will be supported in the future if they aren’t
already

snelweg, 2024-07-20:
They are cheap too, the board cost me less then if I just bought all the connectors!
https://aliexpress.com/item/1005003736312354.html

snelweg, 2024-07-20:
looks simple, will try in a few hours.

```
void setup() {

  Serial.begin(115200);

  delay(3000);

  kc868.begin(115200);//open the kc868  first

  kc868.setReadMode(0);//mode, 0 :default, Query directly; 1: Query all switch state every 2s 

  kc868.setSwitchChangeHook(SwitchUpdate);//set switch change hook

  kc868.setSensorChangeHook(SensorUpdate);//set sensor change hook

  Serial.println("Write Switch 1 on.");
  kc868.writeSwitch(1,1);

  int ret = kc868.readSwitch(1);
  Serial.printf("The state of Switch 1 is %d .\n",ret);

  Serial.println("Delay 1 second.");
  delay(1000);

  Serial.println("Write Switch 1 off.");
  kc868.writeSwitch(1,0);

  ret = kc868.readSwitch(1);
  Serial.printf("The state of Switch 1 is %d .\n",ret);
  

}
```

AlexmagToast, 2024-07-21:
oh wow they are cool indeed. 

I guess once we know which Pin is which IO we could flash the Arduino firmware onto it easily.

snelweg, 2024-07-22:
> oh wow they are cool indeed.
> 
> I guess once we know which Pin is which IO we could flash the Arduino firmware onto it easily.

It's not that easy.
The board uses pc8574's to get the large number of in and outputs. So we have to use the `kc868.begin(115200);//open the kc868  first` driver. But that will work with an edit of your source code.

snelweg, 2024-07-22:
> oh wow they are cool indeed.
> 
> I guess once we know which Pin is which IO we could flash the Arduino firmware onto it easily.

```
#define ANALOG_A1  36
#define ANALOG_A2  34
#define ANALOG_A3  35
#define ANALOG_A4  39

IIC SDA:4
IIC SCL:5

Relay_IIC_address 0x24
Relay_IIC_address 0x25

Input_IIC_address 0x21
Input_IIC_address 0x22

DS18B20/DHT11/DHT21/LED strip -1: 32
DS18B20/DHT11/DHT21/LED strip -2: 33
DS18B20/DHT11/DHT21/LED strip -2: 14

RF433MHz wireless receiver: 2
RF433MHz wireless sender: 15

Ethernet (LAN8720) I/O define:

#define ETH_ADDR        0
#define ETH_POWER_PIN  -1
#define ETH_MDC_PIN    23
#define ETH_MDIO_PIN  18
#define ETH_TYPE      ETH_PHY_LAN8720
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT

RS485:
RXD:16
TXD:13
```