AlexmagToast/LinuxCNC_ArduinoConnector #26: Execv permission denied

paulsvc, 2024-09-05:
Hello,

Thank you for the work, this help me a lot in my retrofit :)

I run into an issue when I try to run the command loadusr arduino-connector
I get the following error : ececv(arduino-connector) permission denied

I verified the chmod and it's executable. I had to modify the python path to /usr/bin instead of usr/bin/env but made no other change.

If I start halrun with sudo, I get the same issue and I don't understand why.
Linuxcnc 2.9.2 from the official iso

Thank you for your help

AlexmagToast, 2024-10-16:
Hi, Sorry for the late reply, did you fix your issue by now? 
Which Python version are you running?

closed 2024-11-04 by AlexmagToast

paulsvc, 2024-11-04:
Here is the solution
Loadusr python3 /usr/bin/arduino-connector