AlexmagToast/LinuxCNC_ArduinoConnector #27: Delay on outputs

bladekel, 2024-10-16:
First of all thats a good job, congrats...

I have been testing your work for 2 days.
There is no problem on reading inputs but there is a big delay on outputs, its about 2 seconds...
I checked it and found that the problem is not on arduino side. The serial data comes to arduino with 2seconds delay(only output). The input information is coming without a delay (at least visibly).

Is this normal or can I fix this?

AlexmagToast, 2024-10-16:
2 Seconds of a delay is definetly to much. Can you please provide your py file so i can check your configuration?

Artem-Viveritsa, 2024-10-18:
> Is this normal or can I fix this?

By default the timeout was 1 sec, I reduced it this way and it became much more responsive.

`arduino = serial.Serial(connection, 115200, timeout=0.1, xonxoff=False, rtscts=False, dsrdtr=True)`

KennethThompson, 2024-10-18:
> > Is this normal or can I fix this?
> 
> By default the timeout was 1 sec, I reduced it this way and it became much more responsive.
> 
> `arduino = serial.Serial(connection, 115200, timeout=0.1, xonxoff=False, rtscts=False, dsrdtr=True)`
> 
> Possible values for the parameter timeout which controls the behavior of [read()](https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.read):
> 
> `timeout = None`: wait forever / until requested number of bytes are received `timeout = 0`: non-blocking mode, return immediately in any case, returning zero or more, up to the requested number of bytes `timeout = x`: set timeout to x seconds (float allowed) returns immediately when the requested number of bytes are available, otherwise wait until the timeout expires and return all bytes that were received until then.

The default was set to something quite high given some Arduino's have limited resources.  If your Arduino does not lose connection randomly with a timeout of .1, then you are good to go.

bladekel, 2024-10-19:
Thanks for your information.

I found that the delay is caused by timeout after I wrote to you.

I decrease that timeout to 0.2 and now it is more acceptable.

Again thanks to you for your great work....
________________________________
Gönderen: Ken Thompson ***@***.***>
Gönderildi: 18 Ekim 2024 Cuma 21:19
Kime: AlexmagToast/LinuxCNC_ArduinoConnector ***@***.***>
Bilgi: Çağrı Cihanoğlu ***@***.***>; Author ***@***.***>
Konu: Re: [AlexmagToast/LinuxCNC_ArduinoConnector] Delay on outputs (Issue #27)

Is this normal or can I fix this?

By default the timeout was 1 sec, I reduced it this way and it became much more responsive.

arduino = serial.Serial(connection, 115200, timeout=0.1, xonxoff=False, rtscts=False, dsrdtr=True)

Possible values for the parameter timeout which controls the behavior of read()<https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.read>:

timeout = None: wait forever / until requested number of bytes are received timeout = 0: non-blocking mode, return immediately in any case, returning zero or more, up to the requested number of bytes timeout = x: set timeout to x seconds (float allowed) returns immediately when the requested number of bytes are available, otherwise wait until the timeout expires and return all bytes that were received until then.

The default was set to something quite high given some Arduino's have limited resources. If your Arduino does not lose connection randomly with a timeout of .1, then you are good to go.

ShadeTechnik, 2024-12-23:
> The default was set to something quite high given some Arduino's have limited resources. If your Arduino does not lose connection randomly with a timeout of .1, then you are good to go.

This is interesting. I pretty much gave up on this project because of the issues with lag.....it was pretty bad. If all it actually was this timeout setting it may be worth testing again. I did 2 versions of a breakout board for this, 1 used an Arduino Mega, the other used a Teensy 4.1. I made the Teensy version because the Arduino version was so slow. There was some issue with my Teensy PCB and it didn't improve anything anyway so I pretty much abandoned it. I'd wonder if I could ude the Teensy, decrease this timeout even further or set it to None, increase the serial baud rate, and likely use UART rather than USB and what the performance might be like in that case.

AlexmagToast, 2024-12-23:
I could not reproduce the Lag on my Test-equipment. Please let us know when you test it again. Also how much of lag do you think is acceptable?